diydrones/MinimOSD-Extra #4: position map

borges2003xx, 2015-02-13:
Great job! it Should be possible having superimpressed to horizont a simbolic map position of plane (or copter) with orientation respect the home  aka in u-osd2 for example?

|-------------------------------------------------- |
|                                          <                   |
|                                                               |
| =========          O                             |  
|                      =========                     |
|                                           ========  |
|-------------------------------------------------  |

sorry for too simple drawing
thanx in advance

iskess, 2015-02-13:
I don't really see how that would be more useful for the pilot than the simple home arrow. Unless the waypoints were also displayed, then that would be very helpful. 
Obviously space is too limited on this board for this.

Elios007, 2015-05-09:
Hi

I don't know if it the same feature that i am looking for. We need a top view radar of the plane/copter around the home on the OSD.

the existing arrow is nice for FPV but also it is important to know where the plane is On the OSD.

Elios007, 2015-05-09:
It is the same as vectorpro OSD or Fy41 OSD and also MFD AP.

I know there is no place on the existing board but there are 2 new boards which can work with the minim osd extra.