diydrones/MinimOSD-Extra #6: Buddy heading

Daniel-1276, 2015-04-14:
Would it may be possible to feed the OSD with mavlink GPS data from a second Telemetry  adapter (different ID)  to integrate navigation/heading arrow guiding to a buddy that is flying with you? That would so awesome, I only know this feature from the myflydream antenna tracker.

iskess, 2015-04-14:
It could be displayed on an OSD or GCS in a radar type view. 
Great for situational awareness.

Daniel-1276, 2015-05-04:
well, I was thinking more of an additional "home arrow"  pointing to your wing man. to make that happen, minimosd would have to receive additional mavlink GPS coordinates, sent via 3DR radio of wing man and received by a second 3DR radio on the following plane... somehow maybe minimosd board could receive both information and calculate direction, just like a "moving home position". Anyone who has an idea how this may could work?