diydrones/MinimOSD-Extra #8: Please add support for showing elevation above terrain

marcmerlin, 2015-04-25:
https://groups.google.com/forum/#!msg/drones-discuss/abJ9fhBbpV0/XUc46rdy2uwJ
Tridge says
"It would be good if the OSD also displayed 
TERRAIN_REPORT.current_height, so OSD uses know their terrain 
height. That value is available even if you are not following the 
terrain."

cmavr8, 2015-04-25:
It would be easier for people to understand what is being requested and motivation, if there was an explanation about it here. 

So, is this gonna be used to debug terrain following?

marcmerlin, 2015-04-25:
Fair question: my plan was to use it for both
1) debug terrain following
2) stay below a specific AGL altitude if needed. Right now I use height above takeoff point, but if I go fly in the hills, that altitude isn't useful or relevant.

Because space is limited, I can see how I would only use 2 altitudes instead of 3: Heigh Above Takeoff Point and Height Above Terrain. I could drop absolute altitude, although if I can have all 3, even better :)

iskess, 2015-04-25:
+1, this would be excellent. Like a poor man's Radio Altimeter. Well, not really, but it can certainly help situational awareness to know your height above the ground below you.

Elios007, 2015-05-09:
HI, 

I am wondering why we don't try the other new boards which can work with the MinimOSD extra, they have plenty of space to add tones of features.

LeChuck71, 2015-05-13:
Hi Elios007,

can you give me a hint, which boards are compatible to minimOSD but with more RAM?

Elios007, 2015-05-13:
Hi Lechuck, I thought that witespy micro minimOSD has more memory, but it seems it is the same.
I saw the rembiziOSD board also but I am not very familiar with it.

LeChuck71, 2015-05-13:
Yeah there are so many out there.
My suggestion would be going for a new video overlay chip that has a higher resolution and maybe does the graphics not via ascii text. But I just cant find any "upgrade" to the current chip. There are OSD i.e. Vector tree that have a way higher resolution. 
Do you know of any video chip capable of lets say 640?

marcmerlin, 2015-11-08:
@Elios007 @LeChuck71 I was looking at this again.
I see plenty of new OSD boards out there, I have a one of these http://www.multirotormania.com/osd/694-mrm-minimosd-.html look at this http://www.multirotormania.com/2073-thickbox_default/mrm-minimosd-.jpg

Of course, I cannot use any of the extra pins for direct measuring of bat and bat2. Not super useful if ardupilot will do it for me, but the point is that I'm hopeful there are boards out there that have more RAM and will allow continuing development of minimOSD instead of being stuck here due to lack of RAM.
Did you guys find any?

marcmerlin, 2015-11-08:
Another interesting board: https://github.com/Brotronics/MinimOSD_MK2