diydrones/MinimOSD-Extra #14: Time to modify it for newer Arduino

AndKe, 2016-01-22:
Won't compile on Arduino from the past 2-3 years, due to the ancient fastserial library (which should not be needed anymore anyway?)

I got stuck modifying it for Arduino 1.6.6.   
Is somebody maintaining it ?

0n3-70uch, 2016-01-22:
Hi,

same problem here. I use this library:
https://github.com/DevFor8/Mav2Duplex/tree/master/libraries

The best solution would be without the FastSerial library.

Greetings
Fabian

thecodingfrog, 2016-01-22:
Install Arduino 1.0.5 and it will work fine

AndKe, 2016-01-22:
@0n3-70uch with that FastSerial, what version of Arduino can you use (how new ?)
@thecodingfrog it works with 1.0.6 too, but it's kind of ridiculous to be stuck with it - it does not improve the impression of being abandoned.

0n3-70uch, 2016-01-22:
I use Arduino version 1.6.5.

AndKe, 2016-01-22:
@0n3-70uch   Thanks , do you compile minimOSD-Extra(copter prerelease) or the Mav2Duplex with it ?
reason for asking is that when I try 1.6.6 (insignificant diffrent from 1.6.5 in this case) , it stops at:

In file included from sketch/ArduCam_Max7456.cpp:4:0:
ArduCam_Max7456.h:77: error: expected class-name before '{' token
 class OSD : public BetterStream {
                                 ^
ArduCam_Max7456.h:83: error: 'uint8_t' has not been declared
     void setPanel(uint8_t start_col, uint8_t start_row);
                   ^

0n3-70uch, 2016-01-22:
Yes there is an diffrent from 1.6.5 to 1.6.6. With 1.6.5 everything works fine.

AndKe, 2016-01-22:
oh - thanks.

0n3-70uch, 2016-01-22:
You are welcome :-)