diydrones/MinimOSD-Extra #15: MinimOSD Extra on Pixhawk Telem2 not working correctly.

geekness, 2016-03-09:
Hi, I have recently updated my MinimOSD (3DR version 1.1) to the latest OSDExtra firmware from github, and started to install it on my Pixhawk (AC3.3.3).
I have the solder jumpers on the board bridged so I can power my camera/video tx seperately.

At first I wasn't getting any data through at all, nor was I able to switch screens (3 pos switch), then I changed some settings in MP in the full tree:
SR2_EXT_STAT=2
SR2_EXTRA1=5
SR2_EXTRA2=2
SR2_EXTRA3=3
SR2_PARAMS=0 (Didn't Change)
SR2_POSITION=2
SR2_RAW_CTRL=2 (Didn't Change)
SR2_RAW_SENS=2
SR2_RC_CHAN=5

BRD_SER2_RTSCTS=0

And confirmed that:
SERIAL2_BAUD=57
SERIAL2_PROTOCOL=1

Now I can see some data updating on the screen, but can not switch the screen.
On boot, and with the switch in POS1, the OSD starts off blank, then as I toggle through the switch positions, every position stays as screen 1. It doesn't look like it's changing or switching after that, my 2 OSD screens are very different so I can tell if they were changing.

So I confirm that my Channel 7 3-way switch is working correctly by checking in MP:
POS1=984
POS2=1260
POS3=1686

Still no luck.
Does anybody have any further info for me to get the screen switching to work?

JoshWelsh, 2016-03-09:
I don't remember the values but yours are too low.  POS1 could be something like 1100, POS2 something like 1500 maybe? Perhaps 1600?  and POS3 1850 or so?  You have to play with the ranges, but, clearly the minimOSD software is seeing your switch changes.

geekness, 2016-03-09:
It only sees the switch once though.
If somebody could chime in with the correct values, that'd be great.

geekness, 2016-03-09:
[https://github.com/diydrones/MinimOSD-Extra/wiki/Screen-switching#2-real-switching](url)
> screen 1: 0 to 1233 screen 2: 1233 to 1467 screen off: 1467 to 1701 (or abowe. the abowe range can be used for additional things)

geekness, 2016-03-09:
OK so going through [https://github.com/diydrones/MinimOSD-Extra/blob/master/MinimOsd-Extra_Copter/OSD_Panels.ino](url)
I have found this:
`//Switch mode by value
      if (switch_mode == 0){
        //First panel
        if (ch_raw < 1200 && panel != 0) {
          osd_clear = 1;
          //osd.clear();
          panel = 0;
        }
        //Second panel
        else if (ch_raw >= 1200 && ch_raw <= 1800 && panel != 1) { //second panel
          osd_clear = 1;
          //osd.clear();
          panel = 1;
        }
        //Panel off
        else if (ch_raw > 1800 && panel != npanels) {
          osd_clear = 1;
          //osd.clear();
          panel = npanels; //off panel
        }
      }
      //Rotation switch
      else{
        if (ch_raw > 1200)
          if (osd_switch_time + 1000 < millis()){
            rotatePanel = 1;
            osd_switch_time = millis();`

Which basically says:
if channel < 1200, then panel 0
if channel >= 1200 & <= 1800, then panel 1
if channel > 1800, then panel off

Which is inconsistent with what is in the wiki. This code listed above won't account for any additional things above 1700, as suggested in the wiki. Maybe the wiki is referring to the Beta version.
Ill test out the different channel values tonight.

geekness, 2016-03-09:
Saying all that though, I should still be toggling through panel 0 and 1, but I can't seem to get past panel 0.

geekness, 2016-03-09:
OK, tested it with the correct PWM settings and still no luck
Anymore suggestions?

geekness, 2016-03-10:
I tried changing to switching to ch8, still no change.

Is there anybody that has panel switching working on AC3.3.3?

venom-code, 2022-08-10:
did you find a solution to the issue?, surprisingly not much input given on such an important issues when using pixhawk, minimOSD and multiple cameras