diydrones/MinimOSD-Extra #16: New features

night-ghost, 2016-03-25:
This firmware does not change for years. New features are rejected due to lack of space.
But see - https://github.com/night-ghost/minimosd-extra

features:
* 4 screens instead of 2
* screen updates only in vertical sync so there is no "snow" on the screen anymore
* individual control of sign icon of each panel per screen
* voltages, current & RSSI can be measured on external pins (that can be found on some boards) 
* RSSI values now is 2 bytes and has no 255 limit
* TLOG player in configurator - now settings can be checked without working plane/copter!
* RADAR (like in MiniNAZAosd) and ILS in Horizon, can be enabled individually
* horizon angles can be adjusted, indepentently in PAL and NTSC
* PAL/NTSC can be detected dynamically - now can use different cameras!
* screen offsets via configurator and via Setup screen
* new GPS coords - in one line
* Any RC channel can be translated to output pin (eg. for cameras switch)
* Setup screen! Some parameters can be adjusted without computer - by RC
* HUD borders can be turned off 
* RADAR point can show track (3 points) - requires charset 2.4.1.2 
* Walkera telemetry output (for DEVO RX705/RX707 receivers) now working! 
* Support for UAVtalk protocol
* Support for CleanFlight/MWII protocol (untested) 
* Support for AutoQuad's MAVlink dialect (untested) 
* support for new APM_Plane modes for quad-plane 

And still 2Kbytes of free flash
discussion - http://www.rcgroups.com/forums/showthread.php?t=2591835

cmavr8, 2016-03-26:
Cool! If it works well and is tested enough, why not merge it back to minimosd-extra? :)

night-ghost, 2016-03-26:
It's still not good tested so it needs  enthusiasts. 

Merge back is impossible because of full rewrite of almost all files.

cmavr8, 2016-03-26:
Ok, then once it's well tested, it should supersede the current code right? I mean, if it's better why not adopt it and everyone can enjoy/contribute. :)

night-ghost, 2016-03-27:
You take a decision, I definitely would not mind. 

PS. The new 817 version has automatic detection of port speed.

cmavr8, 2016-03-27:
I'm not a dev here or anything. The main dev(s) must agree that this is the way to go, then someone does the transition.